mainsail-crew/mainsail #239: [FR] Publish latest mainsail using GitHub Pages

ayufan, 2021-03-26:
**Describe the solution you'd like**
The [fluidd](https://github.com/cadriel/fluidd) publishes an http://app.fluidd.xyz/. The Mainsail could do the same: publish latest release as a regular GitHub Pages web-page allowing to easily use mainsail without installing anything. Just open webpage, put an URL to your Moonraker and start using it.

meteyou, 2021-03-26:
since many releases http://my.mainsail.xyz is available.

ayufan, 2021-03-26:
Oh. This is awesome. Why I didn't see it before? Should we make it prominent in `README.md`?

meteyou, 2021-03-26:
Because the docs guy, is finishing the new docs ATM. And no... This was my first idea, but if you use a HTTPS webapp, you have to use a SSL secure websocket (Moonraker). You can open my.mainsail.xyz with HTTPS, but you have to change your websocket with HTTPS proxy or something else on your printer.

ayufan, 2021-03-26:
So, from `https://` we cannot downgrade to `ws://`? Is that correct @meteyou ?

About `https://` => When I use `https://my.mainsail.xyz` I see the invalid certificate.

meteyou, 2021-03-26:
All modern browsers blocking all connections to none secure sockets.

Thx for the info. I will fix it.

meteyou, 2021-03-26:
https://github.com/meteyou/mainsail/commit/39c2b51faaa43047b1de43ea473cf4c8ede15fdd

closed 2021-03-26 by meteyou

ayufan, 2021-04-09:
@meteyou 

> Thx for the info. I will fix it.

Did you had opportunity to fix HTTPS certificate?