mainsail-crew/mainsail #267: [FR]Login screen....

3dhobbies, 2021-04-21:
**Is your feature request related to a problem? Please describe.**
Not a bug or so, basically a security concern when logging in from external domain (internet). i have done a port forward to my specific printer, and cannot find a way to force the mainsailos to ask for login credentials.

**Describe the solution you'd like**
A basic secure login screen. forcing a user to log into the mainsailos when logging on.

**Describe alternatives you've considered**
Cannot find any other way of securing the mainsailos install on my pi

**Additional context**
Add any other context or screenshots about the feature request here.

meteyou, 2021-04-22:
please do not use port forwarding on your printer! no matter if mainsail, octoprint or anything else is running on the printer! none of these systems were ever built for such a use. please use a secure vpn connection to your network.

apparently, some kind of login has now been integrated into moonraker, so i can also integrate it into mainsail, but it is still not a solution to access the printer via the internet.

3dhobbies, 2021-04-22:
Thank you kindly Stefan.

I am already using a secure VPN to my workshop. The added login would be to
keep nosy employees at bay 😃

Willie Le Maitre
IT DirectorThe Nerd Zone (Pty) Ltd
P: (010)8805851M: 0820430170A: 117A Ouklip Rd, Wilropark, 1724
W: https://nerdzone.co.za
E: ***@***.***
￼
Please do not print this email unless it is necessary. Every unprinted
email helps the environment.

JellevanKraaij, 2021-11-19:
I love to see support for the moonraker authentication functionality as well, hope it gets implemented soon! Just switched from fluidd to mainsail, but the authentication is the only thing that I find really missing.

Thanks a lot!!

kubik369, 2022-01-26:
Does it make sense to reinvent the wheel instead of using something like basic auth in nginx? OP is saying that he could not find any other way, but this seems to fit the bill perfectly, since kiauh already sets nginx up and all :)

meteyou, 2022-01-26:
moonraker authentication does have certain advantages. furthermore, it is also nicer integrated in the gui.

dohlin, 2022-02-23:
Switched from Fluidd to Mainsail and this is about the only thing I'm greatly missing.  Definitely a nice to have!

tfjield, 2022-07-17:
Is it possible to use basic auth in nginx?  I've tried it, but after I provide credentials and Mainsail loads, it can't connect to Moonraker.

rroethof, 2022-09-09:
Currently using Fluidd in combo with moonraker and in this network any network listener without a form of authentication is not an option.
Therefore support for the moonraker auth is a much wanted request here.

MabaKalox, 2023-04-22:
@meteyou hi! How can I help adding this feature?

meteyou, 2023-04-22:
@MabaKalox hey! It would be very nice if you had time to finish the implementation! I started the implementation months ago here: https://github.com/mainsail-crew/mainsail/pull/370, but i haven't time to finish it.

The PR draft is very, very, very old. Starting with a new fork is more uncomplicated than updating my old branch.

meteyou, 2023-04-22:
ahh sry. this was my implementation: https://github.com/meteyou/mainsail-dev/tree/feature/authorization

the PR above was from another guy.

MabaKalox, 2023-04-23:
@meteyou Thank you, I will take a look.

Robnex, 2023-08-27:
Hi. what's the status of the authentification implementation?

midikeyboard, 2023-08-29:
> its open on the roadmap. i currently run mainsail behind nginx proxy with a access list to authenticate while having public access. if you need authentication this is the way to go.

midikeyboard, 2023-08-29:
> Hi. what's the status of the authentification implementation?

its open on the roadmap. i currently run mainsail behind nginx proxy with a access list to authenticate while having public access. if you need authentication this is the way to go (for now).

meteyou, 2023-08-29:
@midikeyboard even if the authentication of Moonraker is integrated, this feature is NOT made for public access!!!

synthead, 2024-12-23:
I'm very interested in this feature. Honestly, I'm surprised that auth hasn't been implemented at this point.

agravelot, 2025-01-24:
> > Hi. what's the status of the authentification implementation?
> 
> its open on the roadmap. i currently run mainsail behind nginx proxy with a access list to authenticate while having public access. if you need authentication this is the way to go (for now).

Opening Mainsail to the public isn't the real issue. The real concern lies in having open access to the Moonraker API. Relying solely on trusted clients isn't sufficient either, as spoofing an IP address is relatively easy. It wouldn't take much effort to write a script to locate moonraker in your LAN (or WAN) and leave a cheerful smiley on your build plate. Even worse, having access to the filesystem through moonraker api...