mainsail-crew/mainsail #271: Bad Gateway uploading a gcode file

thclark, 2021-04-25:
## Describe the bug
I'm uploading a large gcode file, and the upload fails. In Chrome devtools javascript console, I see the following error:
```
Failed to load resource: the server responded with a status of 502 (Bad Gateway)
```

I've tried a 450Mb file, then reduced model resolution and tried a 260Mb file, same problem. Will continue reducing.

## Expected behavior
Expected either the file upload works, or for there to be an informative error about why the file can't be uploaded.

## Logfiles
Please upload your Logfiles here or upload them to [pastebin.com](http://pastebin.com) and paste your log links here.

### klippy.log 

Not much exciting in here

[klippy.log](https://github.com/meteyou/mainsail/files/6372057/klippy.log)

### moonraker.log

Long but the probably-important bit is:
```
2021-04-25 14:38:12,420 [http1connection.py:_read_message()] - Malformed HTTP message from 192.168.0.10: Content-Length too long
```

[moonraker.log](https://github.com/meteyou/mainsail/files/6372059/moonraker.log)

meteyou, 2021-04-25:
pls check your nginx max upload sie in `/etc/nginx/sites-available/mainsail`. there should be a value `client_max_body_size 0;` if you installation is a little bit older, it could be missing.

here is an example of a "new" nginx config file: https://github.com/raymondh2/MainsailOS/blob/master/src/modules/mainsail/filesystem/root/etc/nginx/sites-available/mainsail#L24

thclark, 2021-04-25:
Thanks @meteyou - I have a very long print running note but will check and close this once that’s done! :)

meteyou, 2021-05-01:
Have you been able to solve the problem with it?

closed 2021-05-15 by meteyou

HateFindingNames, 2021-06-25:
@meteyou 
Comming from fluidd but had the same issue. Your hint fixed it for me (changes the value from 200M to 0). Thx a lot!