mainsail-crew/mainsail #272: [FR]3 decimal places for babysteps

gmergulhao, 2021-04-30:
Please allow babysteps of .001 or make it parameterizable under settings > interface similar to extruder and movement control.

This allows a much granular fine leveling of bed/first layer.
Already implemented on Fluidd. May help as inspiration ;)

thanks for the great work so far!

zellneralex, 2021-05-01:
The feature to allow the customizing of the baby stepping is controversial discussed internal for months now. The question is more do we want to give the user the feature to set e.g full step values or is the need to prevent the user from there own stupidness higher.

Now let’s talk a little bit about printers, the typical lead screw type printer with a TR 8x2 has an resolution in Z of 0.0025 mm per step. Setting a BabyStep below the resolution of your printer does not make sense at all, as klipper would round the value to the next microstep. Also not each microstep is “equal” in height and have always a slight tolerance. So setting a BabyStep to e.g 0.001 might be looking nice but is in most cases completely nonsense

meteyou, 2021-05-01:
first of all... please do not make comparisons between the two guis. this has led to major problems several times in the past! this is counterproductive.

furthermore, i see a blind extension to 3 decimals as something wrong, because many printers can't even address it and then it leads to more confusion for the user. Customisable makes more sense, of course, with a little validation.

my personal opinion on this topic: 0.001 babysteps is nothing productive. that is 0.5% of a standard layer height and therefore has exactly no visible change of a first layer. every single component of a hobby 3d printer has more deviation.

closed 2021-08-01 by meteyou