mainsail-crew/mainsail #274: After this mornong's update, this is what i get. Recompiled 100 times. Can't get it to work again.

anastazya, 2021-05-01:
Command format mismatch: adxl345_end oid=%c end1_clock=%u end2_clock=%u limit_count=%hu sequence=%hu vs adxl345_end oid=%c end1_time=%u end2_time=%u limit_count=%hu sequence=%hu
This type of error is frequently caused by running an older
version of the firmware on the micro-controller (fix by
recompiling and flashing the firmware).
Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Protocol error connecting to printer

th33xitus, 2021-05-01:
Why exactly do you think that this is a Mainsail issue?

zellneralex, 2021-05-01:
Not a bug only a stupid user … recompile and flash your rpi mcu. Read  https://github.com/KevinOConnor/klipper/blob/master/docs/RPi_microcontroller.md#building-the-micro-controller-code.

meteyou, 2021-05-01:
this issue tracker is only for mainsail issues. you have a problem with klipper and your mcus.

if you still need help with your klipper installation, feel free to join our mainsail discord and ask in the klipper channel. you're sure to find a user there who can help you.

as far as i can read from the headline, you also have an adxl that is connected via the rpi mcu. i think you forgot to flash the rpi mcu it with the new micro-controller firmware.

closed 2021-05-01 by meteyou