mainsail-crew/mainsail #280: [FR] Manual Bed Mesh point probe

soleshoe, 2021-05-13:
It would be nice to be able to probe each bed mesh point individually while generating bed mesh.

For instance, on my printer there are 6 screws to adjust bed level (X5SA Pro). When generating bed mesh I have to wait for the 6 screws probe, each time I need to change one screw value. It would be nice to be able to probe/adjust each screw individualy, then once values seems OK for each screw, launch an entire bed probe.

freasy, 2021-05-13:
The problem with your idea is, that you change the level of every screw, if you change one. So you arent chaning just one position, you always adjust the complete bed. So a complete screw adjust is always the best way.

meteyou, 2021-05-13:
i think you are using the wrong klipper function for your problem. bed mesh is only there to make a " map " of your bed. if you want to level your bed, you should take a closer look at this function.

https://www.klipper3d.org/Config_Reference.html#screws_tilt_adjust

but now apart from that mainsail is just a web interface for klipper. so any functionality would have to be integrated into klipper itself and then we can use it in the gui and klipper currently has no function for reprobe one point. Alternatively, you can simply "misuse" PROBE_ACCURACY for such a measuring.

i am now closing this ticket here because it is currently not implementable and furthermore i don't think it has any added benefit for the bed_mesh function in klipper. however, if you need help to further use klipper or need help setting up the functions, you are welcome to join our discord and ask for it in the klipper channel.

closed 2021-05-13 by meteyou

soleshoe, 2021-05-13:
> The problem with your idea is, that you change the level of every screw, if you change one. So you arent chaning just one position, you always adjust the complete bed. So a complete screw adjust is always the best way.

Actually, when i need to adjust my bed, i do a complete bed probe first, then adjust each screw one by one to correct the bed horizontal position. This way it’s impossible to fine tune each screw because each scew movement has an impact on entire bed - beeing able to adjust repeatedly each screw would have the same inconvenient without the need to probe entire bed at each screw adjustment