mainsail-crew/mainsail #286: UFP handling broken and dont show thumbnails

forlack, 2021-05-24:
**Describe the bug:**\
UFP files show up as UFP files within the gcode section and are not translated to gcode and no thumbnail is attached.  Mainsail, and moonraker both auto updated around the same time as Cura so I dont know if its Cura or Mainsail that made the change.

**Expected behavior:**\
UFP files would become gcode files with a thumnail attached

**Screenshots:**\

**Logfiles:**\

Relevant Log section from moonraker:

2021-05-24 17:59:04,256 [file_manager.py:_process_file_event()] - Inotify file create: gcodes, /home/pi/gcode_files, CE3_xyzCalibration_cube.ufp
2021-05-24 17:59:04,263 [file_manager.py:_process_file_event()] - Inotify writable file closed: /home/pi/gcode_files/CE3_xyzCalibration_cube.ufp
2021-05-24 17:59:04,292 [file_manager.py:_process_file_event()] - Inotify file delete: gcodes, /home/pi/gcode_files, CE3_xyzCalibration_cube.ufp
2021-05-24 17:59:04,420 [file_manager.py:_process_file_event()] - Inotify file create: gcodes, /home/pi/gcode_files, CE3_xyzCalibration_cube.ufp
2021-05-24 17:59:04,592 [file_manager.py:_process_file_event()] - Inotify file delete: gcodes, /home/pi/gcode_files, CE3_xyzCalibration_cube.ufp
2021-05-24 17:59:04,660 [shell_command.py:run_with_response()] - /home/pi/moonraker-env/bin/python: Traceback (most recent call last):
  File "/home/pi/moonraker/scripts/extract_metadata.py", line 754, in <module>
    main(args.path, args.filename, args.ufp)
  File "/home/pi/moonraker/scripts/extract_metadata.py", line 716, in main
    extract_ufp(ufp, file_path)
  File "/home/pi/moonraker/scripts/extract_metadata.py", line 696, in extract_ufp
    with zipfile.ZipFile(ufp_path) as zf:
  File "/usr/lib/python3.7/zipfile.py", line 1222, in __init__
    self._RealGetContents()
  File "/usr/lib/python3.7/zipfile.py", line 1289, in _RealGetContents
    raise BadZipFile("File is not a zip file")
zipfile.BadZipFile: File is not a zip file

021-05-24 17:59:04,660 [shell_command.py:_check_proc_success()] - Command (/home/pi/moonraker-env/bin/python /home/pi/moonraker/scripts/extract_metadata.py -p /home/pi/gcode_files -f "CE3_xyzCalibration_cube.gcode" -u "/home/pi/gcode_files/CE3_xyzCalibration_cube.ufp") exited with return code 1
2021-05-24 17:59:04,843 [file_manager.py:_process_file_event()] - Inotify writable file closed: /home/pi/gcode_files/CE3_xyzCalibration_cube.ufp
2021-05-24 17:59:04,844 [ioloop.py:_run_callback()] - Exception in callback functools.partial(<bound method IOLoop._discard_future_result of <tornado.platform.asyncio.AsyncIOMainLoop object at 0x75d392b0>>, <Task finished coro=<INotifyHandler._process_inotify_events() done, defined at /home/pi/moonraker/moonraker/components/file_manager.py:1179> exception=FileNotFoundError(2, 'No such file or directory')>)
Traceback (most recent call last):
  File "/home/pi/moonraker-env/lib/python3.7/site-packages/tornado/ioloop.py", line 741, in _run_callback
    ret = callback()
  File "/home/pi/moonraker-env/lib/python3.7/site-packages/tornado/ioloop.py", line 765, in _discard_future_result
    future.result()
  File "/home/pi/moonraker/moonraker/components/file_manager.py", line 1186, in _process_inotify_events
    await self._process_file_event(evt, node)
  File "/home/pi/moonraker/moonraker/components/file_manager.py", line 1279, in _process_file_event
    await node.complete_file_write(evt.name)
  File "/home/pi/moonraker/moonraker/components/file_manager.py", line 825, in complete_file_write
    mevt = self.ihdlr.parse_gcode_metadata(file_path)
  File "/home/pi/moonraker/moonraker/components/file_manager.py", line 1123, in parse_gcode_metadata
    path_info = self.file_manager.get_path_info(file_path)
  File "/home/pi/moonraker/moonraker/components/file_manager.py", line 414, in get_path_info
    modified = os.path.getmtime(path)
  File "/home/pi/moonraker-env/lib/python3.7/genericpath.py", line 55, in getmtime
    return os.stat(filename).st_mtime
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/gcode_files/CE3_xyzCalibration_cube.ufp'
2021-05-24 17:59:05,163 [file_manager.py:_process_metadata_update()] - Error running extract_metadata.py
Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/components/file_manager.py", line 1482, in _process_metadata_update
    await self._run_extract_metadata(fname, ufp_path)
  File "/home/pi/moonraker/moonraker/components/file_manager.py", line 1517, in _run_extract_metadata
    result = await scmd.run_with_response(timeout=timeout)
  File "/home/pi/moonraker/moonraker/components/shell_command.py", line 240, in run_with_response
    self.return_code, stdout, stderr)
components.shell_command.ShellCommandError: Error running shell command: '['/home/pi/moonraker-env/bin/python', '/home/pi/moonraker/scripts/extract_metadata.py', '-p', '/home/pi/gcode_files', '-f', 'CE3_xyzCalibration_cube.gcode', '-u', '/home/pi/gcode_files/CE3_xyzCalibration_cube.ufp']'
2021-05-24 17:59:05,658 [shell_command.py:_check_proc_success()] - Command (/home/pi/moonraker-env/bin/python /home/pi/moonraker/scripts/extract_metadata.py -p /home/pi/gcode_files -f "CE3_xyzCalibration_cube.gcode") successfully finished

forlack, 2021-05-24:
Looks like I dont get any metadata from the gcode now?  Ie print time, layers, etc.

meteyou, 2021-05-24:
Mainsail only display Moonraker data. Pls post this on the Moonraker repo.

closed 2021-05-29 by meteyou