mainsail-crew/mainsail #289: [FR] Support Moonraker power API

Staudi14, 2021-05-26:
# Problem
I would like to turn my printer power supply (which is seperated from the RPI power supply) on and off through the Mainsail interface.

# Proposed solution
Add a check to the connection failed menu that looks if [moonraker power](https://moonraker.readthedocs.io/en/latest/web_api/#power-apis) options are available. If a power device is available then display an "on" Button for it which then enables the device.

# Alternatives
My current work around is to send the HTTP request through postman.

meteyou, 2021-05-26:
On the right top corner menu, every Moonraker power device have a switch/button.

Staudi14, 2021-05-26:
Thank you for the response. This solved my problem

closed 2021-05-26 by Staudi14