mainsail-crew/mainsail #291: Axios

whattheflukz, 2021-06-03:
There are issues with uploading larger files (50MB +). They seem to timeout and fail to upload intermittently. 
I think it's worth considering moving away from axios as I believe this to be the cause.

Please see the following links which support this. 

https://github.com/axios/axios/issues/3577

https://forum.duet3d.com/topic/13740/dwc2-uploading-large-gcode-files-results-in-network-error/2

Let me know what you guys think

meteyou, 2021-06-04:
pls check first the nginx settings and moonraker settings. in older versions of this config files, there was a limit or default setting.

this line in the nginx config is to disable the max file size limit:
https://github.com/raymondh2/MainsailOS/blob/master/src/modules/mainsail/filesystem/root/etc/nginx/sites-available/mainsail#L24

and in moonraker, there is also a "limiter":
https://moonraker.readthedocs.io/en/latest/configuration/#server

in moonraker the default limit is 1024mb, so i think your issue is the nginx config. pls check this settings first.

whattheflukz, 2021-06-05:
I think you may be right. I opened this issue on behalf of someone else and I am not getting any more info from them so feel free to close.
 I was under the impression they were uploading directly from the slicer without issue but that appears to not be the case. 
Sorry about that!

meteyou, 2021-06-05:
Np problem! Pls Ping me, if it don't fix the issue, or write here to reopen this...

closed 2021-06-05 by meteyou