mainsail-crew/mainsail #293: [FR] Display currently select Bed Mesh / HeightMap Profile on Dashboard

craiglink, 2021-06-09:
**Is your feature request related to a problem? Please describe.**
It would be nice to know what the selected profile (if any) is in used when viewing the dashboard screen without having to open the heightmap tab

**Describe the solution you'd like**
Display the name of the selected heightmap profile on the dashboard.  Perhaps in Controls or Misc

**Describe alternatives you've considered**
None.

**Additional context**
None

meteyou, 2021-08-01:
you can also use a M117 for display it in the status panel. would this work for you?

zellneralex, 2022-01-21:
Maybe add it to the z-offset panel. Maybe in the header line since then it would be always visible.

PR #1000 (2022-08-06): feat: show current bed mesh profile name in toolhead panel
Fixes #293
closed 2022-08-06 by th33xitus