mainsail-crew/mainsail #294: [FR] Add browser-based slicer support

VinnyCordeiro, 2021-06-11:
That's an idea that appeared on Voron Discord. There's an open source slicer already available called Kiri:Moto, which is hosted here: https://github.com/GridSpace/grid-apps

It is browser-based, but not cloud-based, so the processing is made locally. Also, Kiri:Moto is licensed with the MIT License, which is compatible with GPL, so on that front that shouldn't be a problem.

The support could be added as a module, in order to not bloat the main installation for those who do not want this feature.

meteyou, 2021-06-11:
Thx for you FR. Looks interesting!

PhilippMolitor, 2023-12-03:
I've seen that OrcaSlicer supports being built for the web, there is some docker image by LinuxServer... could be an interesting integration to consider.

meteyou, 2023-12-03:
@PhilippMolitor pls post a link to this information

PhilippMolitor, 2023-12-03:
https://github.com/linuxserver/docker-orcaslicer

Xolverr, 2024-11-01:
Any updates on this FR? Looking to use Kiri:Moto embedded into my web interface as well.

meteyou, 2024-11-01:
@Xolverr no, and right now, i think i will not implement a slicer in Mainsail. If you like Kiri:Moto, you can use it side-by-side and it would only "blow up" Mainsail with only a very very low number of users. furthermore, there is a very strong difference in preference for slicers and i don't think that such good slicers as Orcaslicer or Prusaslicer manage to get the performance into a browser.