mainsail-crew/mainsail #297: [FR] install deps, config for resonance measurement

blalor, 2021-06-20:
**Is your feature request related to a problem? Please describe.**

When following the guide to performance [resonance measurements](https://www.klipper3d.org/Measuring_Resonances.html) I was surprised that SPI was not enabled, and that numpy and matplotlib were not already installed.

**Describe the solution you'd like**

Preinstall the dependencies and enable SPI.

**Describe alternatives you've considered**

Installing them manually on the system is certainly doable, but numpy in particular is a big dependency and slow to build.

**Additional context**

[image]

I suspect that installing `python-numpy` before creating the virtualenv will make for much faster installation, although the venv may need the flag for including site packages.

meteyou, 2021-06-20:
mainsail is just the webinterface. the webinterface will only be execute in your browser, so there are no possibility to "install" this libs.

blalor, 2021-06-20:
Bah. I knew that! The "recommended" way to install is [via mainsailos](https://docs.mainsail.xyz/setup/mainsail-os) and I just clicked the "mainsail on github" link. Sorry!

closed 2021-06-20 by blalor