mainsail-crew/mainsail #313: Mainsailos looses wifi connection if webcam resolution is higher then standard.

xp1ratex, 2021-07-23:
**Describe the bug:**\[image]If i switch webcam resolution to: camera_usb_options="-r 1920x1080 -f 15" in  /boot/mainsail.txt mainsailos starts to lag after a while and looses wifi connectivity, i have to kick it out of the acesspoint for it to come back. If i switch back to auto settings everything seems to work (its been working for 24h atleast)

**Expected behavior:**\
Running the same webcam configuration on octoprint with the same webcam (logitech HD PRO c920) on my other machine and dont have any issues with the wifi.

**Screenshots:**\

[image]
[image]

**Logfiles:**\
Please upload your Logfiles here or upload them to [pastebin.com](http://pastebin.com) and paste your log links here.
- klippy.log
[klippy (1).log](https://github.com/meteyou/mainsail/files/6870584/klippy.1.log)
- moonraker.log
[moonraker (1).log](https://github.com/meteyou/mainsail/files/6870587/moonraker.1.log)

meteyou, 2021-07-30:
mainsail is just the webinterface... this is a raspberry issue. the wifi chip go in sleep mode.

closed 2021-07-30 by meteyou

xp1ratex, 2021-07-30:
The wifi sleepmode is off, something with the mainsailos is diffrent to the octopi since it doesnt loose wifi if i enable higher cam resolution.
[image]
[image]

ichbaum, 2021-10-10:
I have the same problem. The webinterface stopped working. But only now and than.
No problems with the standard resolution.
I never had that problem with octoprint.

meteyou, 2021-10-11:
this is a network issus. octoprint and mainsail are just webinterfaces...