mainsail-crew/mainsail #314: v2.0.0-beta (config files in subfolders can no longer be edited.)

kolbenhans, 2021-07-26:
**Describe the bug:**\
separated config (*.cfg) files in subfolders can no longer be edited.
Error: "filename.cfg cannot be uploaded"

**Expected behavior:**\
files were still editable without problems before the beta.

**Screenshots:**\
https://ibb.co/qj4SYvq

**Logfiles:**\
Please upload your Logfiles here or upload them to [pastebin.com](http://pastebin.com) and paste your log links here.
- klipper.log
there is no info about the editing in klippy.log

- moonraker.log
https://pastebin.com/qTqMx62p

vonwange, 2021-07-28:
Meteyou gave me this command to fix it: "ls -lt /home/pi/klipper_config/printer/*"

It did not work for me, maybe for you.

kolbenhans, 2021-07-28:
This doesn't fix anything but list all the files in the config path with their file system access.
If this helps @meteyou :
```
~/klipper_config/printer_1 $ ls -lt *
-rw-r--r-- 1 pi pi 1004 Jul 27 21:22 moonraker.conf
-rw-r--r-- 1 pi pi 4360 Jul 27 21:10 printer.cfg
-rw-r--r-- 1 pi pi 4368 Jul 27 20:35 printer-20210727_210141.cfg
-rw-r--r-- 1 pi pi 4368 Jul 27 20:34 printer-20210727_203509.cfg
-rw-r--r-- 1 pi pi 4360 Jul 27 20:25 printer-20210727_203434.cfg
-rw-r--r-- 1 pi pi 3937 Jul 27 20:21 printer-20210727_202530.cfg
-rw-r--r-- 1 pi pi 3094 Jul 27 19:41 printer-20210727_202124.cfg
-rw-r--r-- 1 pi pi 1962 Jun 30 08:02 kiauh_macros.cfg

hillsCube:
total 32
-rw-r--r-- 1 pi pi 2255 Jul 27 20:31 macros.cfg
-rw-r--r-- 1 pi pi  518 Jul 27 20:00 fans.cfg
-rw-r--r-- 1 pi pi 2611 Jul 26 16:04 stepper.cfg
-rw-r--r-- 1 pi pi  285 Jul 15 09:32 bltouch.cfg
-rw-r--r-- 1 pi pi  212 Jul 10 22:07 bedmesh.cfg
-rw-r--r-- 1 pi pi  195 Jul 10 21:53 ztilt.cfg
-rw-r--r-- 1 pi pi  208 Jul 10 10:02 temp_adv.cfg
-rw-r--r-- 1 pi pi  341 Jul  5 16:10 heater.cfg

```

meteyou, 2021-07-29:
This is already fixed. Will be part of the next release.
https://github.com/meteyou/mainsail/commit/ac8367021232ffae063b31ba16d29e4f40bdade6

closed 2021-07-29 by meteyou