mainsail-crew/mainsail #316: Web UI in Safari (Mac OS)

TjardF, 2021-08-01:
Hi,
I recently installed Mainsail and tried to access it via Safari, it only loads a Black window.
With Chrome everything works fine.
Is this a known issue? is there any way to solve it or will it be fixed in future version?
[image]

meteyou, 2021-08-01:
pls more informations... browser version numbers and mainsail version number.

i changed some things for older safari support. this should be in the next beta release (will be released in the next days)

meteyou, 2021-08-07:
@TjardF can you check this again?

TjardF, 2021-08-09:
Hi,
I tried again with 1.6, now everything works fine
OS X 10.15.7
Safari 13.1.3

closed 2021-08-09 by TjardF