mainsail-crew/mainsail #320: [FR] Support for Moonraker user authorization

RealDeuce, 2021-08-05:
**Is your feature request related to a problem? Please describe.**
Yes, I have no IP addresses that I trust, so I use user authentication with moonraker.
```
[authorization]
enabled: true
force_logins: True
```
With this configuration, Mainsail appears to be unable to work.

**Describe the solution you'd like**
In the case where a 401 message is received from Moonraker, Mainsail should prompt the user to log in.

**Describe alternatives you've considered**
I've considered and rejected explicitly trusting connections by peer address.

**Additional context**
Add any other context or screenshots about the feature request here.

meteyou, 2021-08-06:
It's on my to-do list... Maybe a feature for v2.1 or v2.2... I think v2.0 have enough features...

meteyou, 2021-08-11:
this FR already exists.. #267

closed 2021-08-11 by meteyou