mainsail-crew/mainsail #331: Temp chart not displaying data or updating

bill-orange, 2021-08-17:
**Describe the bug:: Operation seems normal but the chart has no data as shown

**Expected behavior:  Temp value displays in chart

**Screenshots: 
[image]

**Logfiles:

[Downloads.zip](https://github.com/meteyou/mainsail/files/7003512/Downloads.zip)

meteyou, 2021-08-18:
is this the first load of mainsail? atm, i'm working on it, if a sensor is not in the moonraker history, it is not included in the chart. if you refresh one time and moonraker provide history data for the sensor it should work...

bill-orange, 2021-08-18:
Thanks.  Refresh corrected the problem.

closed 2021-08-18 by bill-orange