mainsail-crew/mainsail #334: Saving mesh in HEIGHTMAP menu results in "Klipper Not Ready HEIGHTMAP Disabled"

bill-orange, 2021-08-21:
**Describe the bug:

When attempting to save a Mesh in the HEIGHTMAP menu an error message is issued "Klipper Not Ready HEIGHTMAP Disabled"  I am not sure if the Mesh was saved or not.  Saving the Mesh in the Dashboard Menu works fine.

**Expected behavior:

Mesh is saved without (likely false) error.  This error only started appearing on 8/20/2021

**Screenshots: N/A

**Logfiles:
Attached as noted:

[Downloads.zip](https://github.com/meteyou/mainsail/files/7024844/Downloads.zip)

meteyou, 2021-08-22:
this is just the warning, during the klipper restart

closed 2021-08-22 by meteyou

dnewell73, 2023-03-09:
Why doesn't the mesh show up as saved?

jjl40, 2024-08-29:
This is confusing. Are you saying just ignore this error message?

meteyou, 2024-08-29:
@jjl40 Mainsail is just a GUI and display the current state from Klipper. SAVE_CONFIG is a Klipper command, which restart Klipper. While restarting, Klipper is a short time not available and during this time, Mainsail cannot display anything from Klipper, because it is not available. so what should i display? just nothing? Or should i display the correct state "Klipper not ready"? because in this situation, i dont know if you just open this page or will maybe Klipper will never start, because of an issue in your config?

jjl40, 2024-08-29:
Stefan

Thank you for the explanation.
This is my first project using Mainsail, on a new control board. So I’m learning a lot.
Your explanation is very clear and I appreciate your taking the time to clear up my confusion.

Cheers

John Lytle