mainsail-crew/mainsail #342: Safari edit Configs

TjardF, 2021-08-29:
**Describe the bug:**\
A clear and concise description of what the bug is.
After clicking on some files, the download in der right corner appears but the pop up didn't open up. Chrome works

Mac OS 10.15.7 (19H1030)
Mainsail v0.9.1-743-g95047784
**Expected behavior:**\
That the Text editor opens, like in Chrome

**Screenshots:**\
[image]
[image]

**Logfiles:**\
[klippy-2.log](https://github.com/meteyou/mainsail/files/7073034/klippy-2.log)
[moonraker.log](https://github.com/meteyou/mainsail/files/7073035/moonraker.log)

meteyou, 2021-09-03:
do you use safari 14.0? i checked it on big sur with safari 14.1 and it work fine. a friend also reported this bug. seens to be a safari 14.0 bug with BLOB. pls try to update your safari.

B0rax, 2021-09-03:
for context, I had the same problem. You can update Safari to 14.1 on Catalina. It's not very obvious how, but there is an update available.

Go to System Settings -> Software-update
It should tell you that there are further updates available (not the Mac OS update itself):
[image]

click on "more information" and follow the instructions.

Then everything should be working.

meteyou, 2021-09-07:
@TjardF did you tested the tip from B0rax?

retroneo, 2021-09-11:
2.0 and 2.0.1 work fine on Catalina for me.

TjardF, 2021-09-12:
@B0rax Trick worked out perfectly, thanks

closed 2021-09-12 by TjardF