mainsail-crew/mainsail #346: Bed Mesh Issues

Jestinweber, 2021-09-02:
I'm currently trying to get Bed Mesh to work with no luck so far. When I try BED_MESH_CALIBRATE I get an error saying "Already in a manual Z Probe. Use ABORT to abort it."
I cant find anywhere how to resolve my issue so I figured I'd try my luck here. I look forward to resolving this issue!

[klippy (9).log](https://github.com/meteyou/mainsail/files/7102204/klippy.9.log)
[moonraker (2).log](https://github.com/meteyou/mainsail/files/7102205/moonraker.2.log)
[image]

meteyou, 2021-09-03:
this is a klipper config error. this have nothing todo with mainsail. if you need klipper help, feel free to join our discord server and ask in the klipper channel.

closed 2021-09-03 by meteyou

Jestinweber, 2021-09-03:
will do. thanks a bunch!