mainsail-crew/mainsail #347: Thumbnail Preview not showing

atharvbadkas, 2021-09-03:
**Describe the bug:**\
Mainsail dont load thumbnail, even when print starts, on main page it shows blank space.

**Expected behavior:**\
Should have shown thumbnail, I also have updated the moonraker dependencies using SSH.

**Screenshots:**\
[image]

**Logfiles:**\
[moonraker.log](https://github.com/meteyou/mainsail/files/7105597/moonraker.log)
[klippy.log](https://github.com/meteyou/mainsail/files/7105599/klippy.log)

I have recently migrated from Marlin and have limited experience with linux systems and R pi stuff.
I'm using cura slicer and cura2moonraker plugin.
thumbnail gets created in .thumbs folder but it doesnt apply.

meteyou, 2021-09-04:
pls upload one of the gcode files

atharvbadkas, 2021-09-05:
Hi,
I tried without inserting " , " in name but didnt work.

Heres GCODE ( purposely renamed to TXT to upload here)
[9_Q016_A.txt](https://github.com/meteyou/mainsail/files/7111216/9_Q016_A.txt)

meteyou, 2021-09-08:
thx for the gcode. i think i found your issue. the problem is, that your gcode have only the big thumbnail in it. the next issue is, that you have a thumbnail in the gcode. if you use the cura moonraker plugin, you should upload the UFP file and not the gcode. this file is like a zip file with the files in it. is it possible, that you use the postprocessing to add the thumbnails in your gcode file? if you use it, pls add a second thumbnail with 32x32px.

closed 2021-09-08 by meteyou

atharvbadkas, 2021-09-09:
Thanks, adding second thumbnail size made it work!