mainsail-crew/mainsail #349: Main background not working after 2.0 upgrade

jaydlawrence, 2021-09-03:
**Describe the bug:**\
I use the theming options of mainsail to differentiate my printers with different interfaces.
I used both sidebar-background.jpg and main-background.jpg.
I just updated both my printers to mainsail v2.0.0 and both printers of them seem to not be showing the main background anymore, but do show the sidebar.

**Expected behavior:**\
Usually, when you upload a main-background.jpg, it shows up as the background in the UI, this time it is not.

**Screenshots:**\
[image]

**Logfiles:**\
https://pastebin.com/wPnjq2BV
https://pastebin.com/3SpLm01Q

jaydlawrence, 2021-09-04:
I am not very familiar with Vue.js, I usually work with React.
But I had a poke around and tried to get to the bottom of why this is happening.
I am not exactly sure why, but the problem is that the `mainBackground` is getting recomputed here:
https://github.com/jaydlawrence/mainsail/blob/master/src/App.vue#L71

```
    get mainBackground() {
        return this.$store.getters["files/getMainBackground"]
    }
```

When the page loads, the file value is null so is computed to `transparent`.
But once the page has finished loading it recomputes to the file path.

But it is not reloading the new value here:
https://github.com/jaydlawrence/mainsail/blob/master/src/App.vue#L26

```
        <v-main id="content" :style="{ background: mainBackground }">
```

So it seems like some sort of style caching thing.

To test this, I make a random extra property to that div like:

```
<v-main id="content" :style="{ background: mainBackground }" :blah="mainBackground ">
```

And on the initial load, they are both set to `transparent`, but once the app has had time to load and resolve the files, it will replace the blah property value but not the style.

I hope this is helpful in getting to the bottom of it.
Sorry that I couldn't be more helpful.

If you don't have time to look into it, but if you have some hints as to why this is happening I can try and have another look with your suggestion in mind.

Cheers

meteyou, 2021-09-04:
fixed the bug. thx for the report!

closed 2021-09-04 by meteyou

nobimoto1, 2022-04-15:
Hi
I have almost the same problem, I want change my sidebar-background but the sidebar-background is still the old. can you help me?
[image]

meteyou, 2022-04-15:
@nobimoto1 the background image is still in your browser cache and will not load again.

nobimoto1, 2022-04-15:
than you for a fast answer, how can I delete the old background

nobimoto1, 2022-04-15:
thanks for help, it´s fixed yet