mainsail-crew/mainsail #352: Retract and Extrude buttons gray after setting min_extrude_temp

kernelsandoz, 2021-09-08:
Retract and Exclude buttons remain grayed out after min_extrude_temp is exceeded and browser refreshed.

Expect these buttons to brighten up when extruder is above min_extrude_temp.

The extruder works, I can start a test print. I have also tried homing first.

[image]

[klippy (2).log](https://github.com/meteyou/mainsail/files/7127613/klippy.2.log)
[moonraker.log](https://github.com/meteyou/mainsail/files/7127623/moonraker.log)

I have already calibrated the extruder so I only need these feature for loading filament.

I notice this has been fixed in previous releases. Am I doing something wrong ?

Thanks for the software, it is neat.

meteyou, 2021-09-08:
Pls post a screenshot of your tempchart. You set your min_extruder_temp to 90°C but you never reach this. the highest temp i could find in your log was 21.7°C. you have mulptile problems with your heater, that it does not heat, if you set the target temp to 190°C.

it looks like a klipper config / hardware issue for me...

kernelsandoz, 2021-09-08:
Thanks for the prompt reply.

[image]

Also I forget to mention.

http://hypercube.local/printer/objects/query?extruder

{"result": {"status": {"extruder": {"pressure_advance": 0.0, "target": 200.0, "smooth_time": 0.04, "power": 0.5357398301943612, "temperature": 200.1035816703706}}, "eventtime": 497.96114882}}

No mention of can.extrude True or False.

The 90C in klippy.log was the default set in WrecklabOS the distribution for my PrintHat2. I have also tried 170,0 and 20.

To be clear filament melts and comes out of the nozzle when I start a benchy.

[moonraker (1).log](https://github.com/meteyou/mainsail/files/7131840/moonraker.1.log)

[klippy.log](https://github.com/meteyou/mainsail/files/7131808/klippy.log)

meteyou, 2021-09-08:
ok. now i checked your klipper version. you have v0.9.1-499 but the value `can_extrude` was added in v0.9.1-572.

pls update klipper to fix this issue.

meteyou, 2021-09-14:
@kernelsandoz any updates to this issue?

kernelsandoz, 2021-09-14:
I raised the issue  with the vendor of my controller board, Wrecklab. The
controller board PrintHat2 is a Raspberry Pi hat and uses the gpio for usb
comms.
It is therefore easier to use their Pi distribution which includes config
files and scripts to facilitate the usb comms and flashing of their board.
wrecklabOS uses its own REPO for klipper and is keeping klipper at a lower
revision.

I can manage without retract/extrude buttons for a while and will wait for
the next version of wrecklabOS which will update klipper.

Thanks for your help.

closed 2021-10-31 by meteyou

kernelsandoz, 2021-10-31:
All good now, thanks very much.
WrecklabOS now has klipper 0.10.