mainsail-crew/mainsail #357: [FR] Link to timelapse video from mainsail UI

afbenevides, 2021-09-17:
I am using this code to make a time lapse video of the prints, but the output file is not available through the mainsail UI. code available here: https://github.com/FrYakaTKoP/moonraker/blob/dev-timelapse/moonraker/components/timelapse.py

Would be interesting if Mainsail could provide a link to the video for the user to download when needed. 

One solution would be to add a video_link column to the "Print History" table. Maybee more complexe, needs a way to know when the video is available, which print it is linked to and then add link, needs also to manage history/video deletion for space..

A simpler first solution, would be to provide a new path in "config Files" "root" drop menu, where we could have a list of available video renders, from which user can choose the one they want to download ( they could also delete obsolete old ones) With that solution, it is also possible to not have the timelapse script defined and setup in our slicers, without creating UI problems... the list will just be empty if no video are available.

FrYakaTKoP, 2021-09-17:
@afbenevides 
We working on Mainsail integration of Timelapse right now. It is already a confirmed feature v2.1!

In the mean time if you have a recent version of the timelapse plugin installed, you already can download the timelapse within mainsail. Your suggestion to provide the timelapse root in the config selection is already in v2.0.x of mainsail, but it's currently limited to dowload the files only. So no delete atm, because i can't subscribe a writeable directory within the component (this will change in near future after rework of file_manager.py is done in moonraker). You could in theory add the timelapse to full access roots but i highly recommend not to do, because you would loose update functionality of moonraker within mainsail.

If you like to receive some teasers and sneak peaks of upcoming features of mainsail, please consider joining our discord server:
https://discord.com/mainsail (alternate link: https://discord.gg/skWTwTD )

afbenevides, 2021-09-17:
@FrYakaTKoP ok thanks I will try in the 2.0.x version! Also I just logged to discord to follow. Thanks a lot!

meteyou, 2021-12-04:
this is already merged in mainsail. if you want to test it, you can update to the newest beta release. thx for your FR.

closed 2021-12-04 by meteyou