mainsail-crew/mainsail #359: Version mismatch...did I screw up somehow?

AaronHerrick, 2021-09-19:
**Describe the bug:**\
The Mainsail version in the lower left does not update from 1.6.0 to 2.0.1

**Expected behavior:**\
I expect it to update the version

**Screenshots:**\
[image]

**Logfiles:**\
Please upload your Logfiles here or upload them to [pastebin.com](http://pastebin.com) and paste your log links here.
- klippy.log
- moonraker.log
I can't do that for ... reasons.  but let me know if you really need something from them

AaronHerrick, 2021-09-20:
closing, the value did update a few hours later.  Could it be because I had 2 different computers viewing the website?

closed 2021-09-20 by AaronHerrick

meteyou, 2021-09-20:
yep. when you use 2 pcs (2 differnt browsers), your browser will cache mainsail. on the pc/browser you update mainsail, ther will be a "force reload" and it will update the static files. on the other pc, you have to reload without cache. for example

cmd+shift+r on mac
ctrl+shift-F5 on pc