mainsail-crew/mainsail #360: [FR] Confirm Changes on Mobile

awickham10, 2021-09-21:
**Is your feature request related to a problem? Please describe.**
Sometimes when using the mobile interface I will accidentally change speed/extrusion/fan speeds when scrolling through the interface.

**Describe the solution you'd like**
It would be nice to see a confirmation popup when this happens. Are you sure you'd like to change the fan speed from 100% to 2%?

**Describe alternatives you've considered**
Disabling edit and forcing the user to click a button to edit seems too cumbersome.

meteyou, 2021-09-21:
we have discussed this several times. mostly on the discord server. there just hasn't been time to implement it yet.

th33xitus, 2021-11-26:
The feature was recently added to Mainsail v2.1.0-beta1 and also includes a user-configurable, timer-controlled re-lock function.

closed 2021-11-26 by th33xitus