mainsail-crew/mainsail #361: [FR] default UI configuration via config file

simonkuehling, 2021-09-21:
**Is your feature request related to a problem? Please describe.**
When deploying a custom built Klipper/Moonraker/Mainsail OS to a number of machines (that each run on a Raspi with a touch monitor attached), I miss having an option to pre-configure the UI through a config file - as all configuration in the interface is stored in the moonraker DB these days (right?)
This applies to the selection and order of the widgets as well as jogging speeds etc. for example.

**Describe the solution you'd like**
If I have researched correctly, the GUI settings have been stored in gui.json inside the sdcard directory before using the moonraker DB. How about re-introducing that file as a default config to populate the moonraker DB upon first start?

**Describe alternatives you've considered**
Have not come up with a workaroud yet - but, I am all ears if I‘m missing something!

meteyou, 2021-09-21:
hey! thx for your FR. gui.json was the old store of the gui settings. now all moved to the moonraker database.

but i already thought about such a function, that you can pass gui settings with an image and load them into the moonraker db the first time you load the gui.

you are welcome to come to our discord and we can discuss how we can implement such a solution. i can also send you test build then.

meteyou, 2021-12-19:
hey @simonkuehling,

i created a PR for this feature (https://github.com/mainsail-crew/mainsail/pull/476). do you have time to test it?

you can config your mainsail to your preferences and then create a backup. this backup can be used for a default mainsail settings file for a rpi image. you only have to add this file in /config/.theme/default.json. this will be loaded the first time, when you try to load mainsail and there is no mainsail db namespace in moonraker.

you can find more informations of the structure of this file here: https://github.com/mainsail-crew/mainsail/blob/96156b985e9d6fac2fd9651db3f3871ec8ededea/src/store/gui/types.ts#L6

closed 2021-12-19 by meteyou