mainsail-crew/mainsail #365: Blank page on iOS 9.3.6 (iPad Mini v1)

dJOS1475, 2021-09-29:
**Describe the bug:**\
I was hoping to use my old iPad Mini v1 as a Klipper Screen (perfect size and decent screen) but connecting to Mainsail results in a blank page regardless of which browser I try. It does recognise the connection but nothing loads.

**Expected behavior:**\
Mainsail to load and work normally. I know iOS 9.3.6 is ancient but these old iPads are ideal for this role.

**Screenshots:**\
[image]

**Logfiles:**\
There's nothing relevant in the logs

lixxbox, 2021-09-29:
Hello @dJOS1475, thank you for your request. 

We have already tried to optimize the browser support as much as possible. Even though Vuetify (the base for Mainsail) specifies Safari 9 as the bare minimum, we have not been able to reach that. Mainsail uses other components besides Vuetify, that may rely on more recent browser features, and can currently be used on iOS10 (afaik) and later.
We will start testing browser support soon and set some minimum requirements.

If you or anyone else has a fix, feel free to contact us.

dJOS1475, 2021-09-29:
Thanks, I can’t code unfortunately or I’d take a swing myself.