mainsail-crew/mainsail #402: Where are the RasPi Images?

altimmons, 2021-11-03:
**Describe the bug:**\
Images for raspi are unavailable.  Reviewing Tutorials online show they were at one time in releases. 
Documentation goes into great detail on how to flash, without anything TO flash

**Expected behavior:**\
A clear and concise description of what you expected to happen.
The Recommended Install method of the Image should be available, or some such

altimmons, 2021-11-03:
found it, via discord-
some hint to check https://github.com/raymondh2/MainsailOS/releases would be ideal, via ReadTheDocs or in README.md

meteyou, 2021-11-03:
we use `Jekyll` docs and it is available via this url: [https://docs.mainsail.xyz](https://docs.mainsail.xyz). this URL is linked multiple times in [https://github.com/meteyou/mainsail/blob/develop/README.md](https://github.com/meteyou/mainsail/blob/develop/README.md)

closed 2021-11-03 by meteyou