mainsail-crew/mainsail #404: [FR] Export Print History as .CSV

troypouliot, 2021-11-05:
**Is your feature request related to a problem? Please describe.**
Yes, I keep a single excel log for all of my printers to track total hours. Every month or every other month I will export a .csv file from the Octoprint plugin "Print Job History". This exports a new row for each print job with all the relevant info (pretty much everything listed in the Mainsail print history).
This makes it a breeze to copy/paste the needed rows to my log file to update the cumulative printer time.

**Describe the solution you'd like**
A button to export as .CSV all or selected print jobs with all relevant info in the print history tab.

**Describe alternatives you've considered**
Going back to Octoprint ;(

**Additional context**
This very important feature is also missing from the Fluidd platform. I chose to request it here as this project looks to be a bit more active with development.

Thanks!

[image]

meteyou, 2021-11-05:
thx for your request. can you upload an example with 1-2 jobs in the CSV file? this feature should not be a problem to implement it.

troypouliot, 2021-11-05:
Here you go-

[OctoprintPrintJobHistory (12).csv](https://github.com/meteyou/mainsail/files/7486513/OctoprintPrintJobHistory.12.csv)

Not all columns are required for me, but other may find use for them. 

My personal record keeping log uses the columns;
1. Print Result - Used to help filter out failed prints (I also look for print time to be < 10 mins)
2. Start Datetime - for sorting
3. Duration - Obvious :)

Thanks for the quick response! Keep up the great work.

meteyou, 2021-11-05:
np. how do you sort your printers? would it be nice to have one column with the printername?

troypouliot, 2021-11-05:
It wouldn't hurt, but I don't know that it is necessary either.
Right now, each printer runs it's own Octoprint instance, so when it is time to log hours I just end up with a seperate csv file for each printer and handle them individually. 

I only have one printer running Mainsail right now as I test it out and make sure it has what I need before switching over the others.

So, yeah now that I think about it more, it would be useful to have the printer name _somewhere_. Either a column that includes this. 
Or, since I think I will be doing the same thing with running seperate instances of Mainsail for each printer, maybe include the printer name in the filename of the csv export? something like [printer_name]-print_history-[export date].csv.
That would make it easy to keep all of them together in a directory.

meteyou, 2021-11-05:
thx for the feedback. i added this feature to my backlog, but i can't say exactly when i will be able to integrate this feature.

PR #675 (2022-02-28): feat: export print history as CSV
This PR add a function to export selected or all jobs to an CSV file.

fix: #404
closed 2022-02-28 by meteyou