mainsail-crew/mainsail #407: [FR] Would it be possible to have Mainsail display the Gcode thumbnail in the File-viewer?

srbjessen, 2021-11-11:
I haven't been on Github for long, so not sure about "correct approach", Please advise if I did it wrong. 

I would love to have the option of seeing the thumbnail of a gcode in the fileviewer. Perhaps just as a pop-up? That way it doesn't ruin the current design/setup. 

Would also be nice if it was possible to see it on the dashboard while printing. I have a TFT70 display with KlipperScreen, but Prusaslicer makes all thumbnails orange, making it hard to see some of the details. 

Thank you, only used mainsail for a week, but already loving it ... 👍

lixxbox, 2021-11-11:
this is already possible. you can find instructions here:
https://docs.mainsail.xyz/quicktips/thumbnails

srbjessen, 2021-11-11:
Well color me embarrassed.... thank you. I will enhance my google-fu for the future....

closed 2021-11-11 by srbjessen