mainsail-crew/mainsail #409: [FR] Export the console messages (commands + results) into a text file

nesdnuma, 2021-11-14:
Hello,

It would be nice to be able to export the text of the console, for instance when performing some commands with the help of some person who is remote.

Also, it would be nice if this text gathered the commands launched from several devices. So having this saving feature from the RPI side rather than from the client side would be even better (I often launch some commands from my computer and some from my phone).

lixxbox, 2022-01-17:
Hello,
thank you for your suggestion. The second part would have to be implemented in moonraker first. Maybe you should create a FR there and link it here. Then we can have a closer look at it. 

I have set the status to "blocked" for the time being.

nesdnuma, 2022-01-17:
Hello,

I have just made a FR on Moonraker dev: [https://github.com/Arksine/moonraker/issues/335](https://github.com/Arksine/moonraker/issues/335)

meteyou, 2023-03-13:
We can reopen this FR, when this function is implemented in Moonraker.

closed 2023-03-13 by meteyou