mainsail-crew/mainsail #410: [FR]Reset to G90 or show G91 when posting error "Move out of range" in console, when moving with controls panel

Tharnvar, 2021-11-14:
**Is your feature request related to a problem? Please describe.**
When moving out of range with the control panel an error message is shown in console.
Even though the printer didn't move, G91 was set without displaying G91 in console.

[image]

The following control panel inputs where without G91 since it was set when the error was sent.
Respectively a manual G1 command will be in relative positioning as well.

**Describe the solution you'd like**
Don't set G91 when moving out of range/Reset to G90 if it was set previously

**Describe alternatives you've considered**
Show G91 before error in console/Add a warning in the error, that G91 was set

**Additional context**
Add any other context or screenshots about the feature request here.

meteyou, 2021-11-14:
mainsail is just a web gui and just display the error. the switch between G90 and G91 have to be a klipper function... all console lines are only the output of klipper.

Tharnvar, 2021-11-14:
Oh, please excuse my lack of knowledge, I will move it to the appropriate place.
Thanks for the fast response!

meteyou, 2021-11-14:
ahh ok... a other mainsail crew member have explained the issue to me. so...

the issue is, that mainsail send all 3 commands at the same time and klipper stop executing the tribble command in the middle, thats why the change back to G91 will not be executed.

Tharnvar, 2021-11-14:
Ah the false spontaneous reopening. Can't think of a solution from mainsail side, will move it to klipper

closed 2021-11-14 by Tharnvar

meteyou, 2021-11-14:
i just think a also good solution would be to just change to relative mode and dont switch back to absolute. its not really necessary to switch back. so the absolute mode gcode would be no problem ever more..

Tharnvar, 2021-11-14:
True then one would see the G91 command in console and expect the relative positioning after using the controls panel.

meteyou, 2021-11-14:
but you will also see the correct value if you check the status panel. so if you send the move command and it cancel it with "move out of range", the mode in the status panel should be the right one, because klipper send every time the new value, if it really change.

Tharnvar, 2021-11-14:
Yes the status page correctly shows the relative mode when the error is posted.
I just made the novice mistake no to check the status screen at all when manually inputing movements afterwards.
(Nothing happened, since I don't have any obstacles in the boundaries yet and "out of range" would still trigger when trying to leave the boundaries)