mainsail-crew/mainsail #416: Editor Syntax Highlighting Error

smackoz, 2021-11-19:
If I click on say the printer.cfg file and open it for editing and press [enter] on a blank line (or anywhere) in the document then offscreen text will be displayed as white text rather than syntax highlighted. If a press enter again when the unhighlighted text is on screen the text is syntax highlighted correctly as expected.

All text should be syntax highlighted irrespective of how many times [enter] is pressed on a blank line in the file being edited.

[image]

[image]

Have repeated this behaviour with Brave, Firefox and Dissenter web browsers.

**Logfiles:**\
n/a

th33xitus, 2021-11-22:
Hi,
we are aware of that issue. Currently we don't exactly know what is triggering this issue but we are in the process of investigating and therefore resolving the issue.

smackoz, 2021-11-22:
> Hi, we are aware of that issue. Currently we don't exactly know what is triggering this issue but we are in the process of investigating and therefore resolving the issue.

I remember seeing a comment about this being unique to chrome based web browsers and something to do with a bug in the way it displays data line-by-line (I think). Apparently until you changed the setting it didn't behave as expected. I have done a search through the Brave browser settings and so far can find no reference to the setting mentioned> This might help you to track down the issue. If what they say is correct then the issue is entirely with the browser rather than mainsail in any case.

meteyou, 2022-01-07:
should be fixed with this PR #462

closed 2022-01-07 by meteyou