mainsail-crew/mainsail #474: Mainsail incorrectly calls "max_accel_to_decel" Decelleration in UX

psiberfunk, 2021-12-17:
**Describe the bug:**\
In the UX for mainsail, in the machine tab , mainsail represents the config value _max_accel_to_decel_ as "Deceleration"
[image]

This is not just technically incorrect, it is misleading/confusion-causing.  As [Klipper explains here ](https://www.klipper3d.org/Config_Reference.html#common-kinematic-settings), this variable is to limit the impact of extreme acceleration-to-deceleration transitions typical during short zig-zag moves and is a "pseudo-acceleration" value that is recommended as ~1/2 of the max acceleration value.  It is more properly understood as a limit on the top speed of really short zig-zag moves which tend to cause the printer to vibrate a lot.

th33xitus, 2021-12-17:
Thanks for pointing out that issue. Yes that's something we have to correct.
I think it is best to stick to the name `Max Accel to Decel` which was given to that value by Klipper then.

meteyou, 2021-12-19:
issue is fixed with this pr: https://github.com/mainsail-crew/mainsail/pull/475

closed 2021-12-19 by meteyou