mainsail-crew/mainsail #479: escape key does not close modals

blalor, 2021-12-20:
**Describe the bug:**\
Pressing the escape key does not dismiss modals, such as the interface settings configuration screen.  The dialog "pulses" but does not close.

**Expected behavior:**\
The dialog should close.

**Screenshots:**\
https://user-images.githubusercontent.com/109915/146761991-05819546-f62f-4288-bb67-975974b4c622.mov

**Logfiles:**\
n/a

steadyjaw, 2022-01-17:
Good day @blalor 
thanks for the report.
This is already fixed and will be available in the next release of Mainsail: #413

Mainsail v2.1 will be released soon. Stay tuned!

steadyjaw, 2022-01-17:
I've closed this issue for now.
Feel free to re-open it, if something is not right in the next release.

closed 2022-01-17 by steadyjaw