mainsail-crew/mainsail #480: G-Code Thumbnails don't get displayed/processed with % in the name

TheLion, 2021-12-24:
**Describe the bug:**\
G-Code Thumbnails don't get displayed/processed with % in the name

**Expected behavior:**\
I expected that the thumbnails would display ;-)

**Screenshots:**\
[image]
The G-Code is exactly the same. Only the filename is different.

**Logfiles:**\
There is nothing in the klippy.log or moonraker.log that has to do with this issue.

meteyou, 2021-12-24:
Already fixed and is included in the current beta.

closed 2021-12-24 by meteyou

9R, 2022-01-19:
This bug is fixed in the G-CODE FILES tab but seems to persists in the DASHBOARD print status view.

Tested with v2.1.0-rc1

[image]

meteyou, 2022-01-19:
@9R fixed after RC1