mainsail-crew/mainsail #489: Temperatures no Table anymore

Devil-of-Chaos, 2021-12-28:
**Describe the bug:**\
In the Temperature Widget the Table off sensors to set Temperature are missing in Current Beta. 
iOS 15 Safari and MacBook latest Chrome Browser

**Expected behavior:**\
There should be the Table width Temperature settings

**Screenshots:**\
[image]

kiwiiikiwiiikiwiii, 2021-12-28:
It works for me. iOS 15.2, safari

[image]

RyanDam, 2021-12-28:
Not shown up for me also

[image]

meteyou, 2021-12-28:
Plus reset your presets via interface settings > general > factory reset and check the presets checkbox

RyanDam, 2021-12-28:
> Pls reset your presets via interface settings > general > factory reset and check the presets checkbox

This works for me

Devil-of-Chaos, 2021-12-28:
Worked thank you maybe there should something in the update script for release

closed 2021-12-28 by Devil-of-Chaos