mainsail-crew/mainsail #504: [FR] neopixel support

meteyou, 2022-01-01:
display neopixel colors and use a color picker to change the color of this neopixel. maybe add a function to group numbers from one line of neopixels.

psiberfunk, 2022-01-05:
Rudimentary inspiration might be drawn from the UX here.. but it is highly constrained by the screen software used on this printer:  https://github.com/CR6Community/Marlin/releases/tag/v2.0.8.1-cr6-community-release-6.1

lbibass, 2022-05-22:
Any update on getting this feature developed?

meteyou, 2022-05-23:
We build up test boards with all kinds of LEDs to test it. Could be possible a feature for v2.3.0. maybe we will make some PRs also to Klipper for better usebillity.

arshillimbani123, 2022-09-06:
> We build up test boards with all kinds of LEDs to test it. Could be possible a feature for v2.3.0. maybe we will make some PRs also to Klipper for better usebillity.

any update in this
?

meteyou, 2022-09-06:
@arshillimbani123 in progress... i linked the PR for this feature.

closed 2022-10-18 by meteyou