mainsail-crew/mainsail #526: Issuetext isnt clear enough

DrFailbucket, 2022-01-08:
It took me around 4h to figure out that it matters if you taype [ **I** nclude] or [ **i** nclude]

The warning about that wasnt clear at all bc, the issuetext it was all lower case so i couldnt notice it directly.

meteyou, 2022-01-09:
pls post a screenshot of the error message to check where the error comes from... if the header is called "klipper error" it depends every time to klipper...

zellneralex, 2022-01-21:
Ok I did a test, and sorry that is complete out of our control.
Error Message seen in Mainsail:
[image]

Config section to generate the error:
[image]

And last but not least the klippy.log:
[image]

The message is generated within klipper and Mainsail only shows what klipper sends as error

closed 2022-01-21 by lixxbox