mainsail-crew/mainsail #533: Emergency Stop button doesn’t appear when Klipper is started

lukasfink1, 2022-01-12:
**Describe the bug:**\
When Mainsail is accessed while Klipper is stopped, the “Emergency Stop” and the “Upload & Print” buttons won’t appear. When Klipper is then started, the page needs to be reloaded for these buttons to appear, while the rest of the UI updates automatically. They also won’t disappear when Klipper is stopped without reload.

The hostname printed next to the Mainsail logo in the top left corner is affected by this too.

**Expected behavior:**\
Those buttons and the hostname should either

- be shown always and independently of the state of the Klipper service or they should
- dynamically appear and disappear depending on the state of the Klipper service without reload.

**Screenshots:**\
Accessing Mainsail while Klipper is stopped:
[image]
After starting Klipper:
[image]
After reloading the Mainsail page:
[image]
After stopping Klipper:
[image]
After reloading the Mainsail page again:
[image]

**Logfiles:**\
[moonraker.log](https://github.com/mainsail-crew/mainsail/files/7858333/moonraker.log)
[klippy.log](https://github.com/mainsail-crew/mainsail/files/7858350/klippy.log)
of the sequence outlined in the Screenshots sections

steadyjaw, 2022-01-13:
Good day,
thanks for the report. I'll have a look on that.

steadyjaw, 2022-01-22:
Fixed in commit:
https://github.com/mainsail-crew/mainsail/commit/674ad36ddb738f4e27b1aac03116f17730319ceb

closed 2022-01-22 by steadyjaw