mainsail-crew/mainsail #534: [FR] Allow user to move panels between dashboard columns in web interface

voidtrance, 2022-01-12:
**Is your feature request related to a problem? Please describe.**
It is currently not possible to move the different panels between the columns on the dashboard. This leads to wasted space because the panels can't be balanced and the user has to scroll to reach panels that could be moved to the second column

**Describe the solution you'd like**
The interface already allows for enabling/disabling the desired panels in each of the columns. However, it currently assigns panels to specific columns. For example, webcam, z offset, controls, macros, etc. are all assigned to the left column and temp. and console the only ones assigned to the right.
Ideally, the user would be able to drag-and-drop panels to columns. This will allow not only for the panels to appear in different columns but also let the user arrange them according to their own preference.

**Describe alternatives you've considered**
Alternatively to implementing drag-and-drop, every panel could be listed in the each of the columns in the Interface Settings -> Dashboard screen. This would allow the user to show the panels that they want in the columns they want.

**Additional context**
None

meteyou, 2022-01-13:
This is already implemented as you have described it...

closed 2022-01-13 by meteyou

voidtrance, 2022-01-14:
Can you please point me to documentation about this? What version is this implemented in. I am currently running v2.0.1 and I cannot move and/or re-arrange the panels at all.

th33xitus, 2022-01-14:
[image]