mainsail-crew/mainsail #548: Bedmesh: deleted meshes are not visible indicated in the UI

zellneralex, 2022-01-19:
**Describe the bug:**\
If you delete a mesh in the UI there is no visible indication that it will be removed after the next save config and also that it is not usable anymore in the current klipper season 

**Expected behavior:**\
Any kind of visualization e.g cross out the name or a small trash can symbol next to the name

**Screenshots:**\
If applicable, add screenshots to help explain your problem.

**Logfiles:**\
Please upload your Logfiles here or upload them to [pastebin.com](http://pastebin.com) and paste your log links here.
not needed

Misterke, 2022-02-15:
This is a consequence of MainSail showing the meshes from the config as currently bed_mesh.py is not offering a way to report the actual available profiles.  Feel free to get more attention for https://github.com/Klipper3d/klipper/pull/5222 as that would allow solving this problem.

meteyou, 2022-02-15:
@Misterke thx for the info and the PR in klipper! with this PR, it would be very easy to fix this issue!

Misterke, 2022-02-16:
> @Misterke thx for the info and the PR in klipper! with this PR, it would be very easy to fix this issue!

The PR has been merged, so Mainsail could now use the actual profiles instead of the static config.

meteyou, 2022-02-18:
fixed with https://github.com/mainsail-crew/mainsail/pull/660

closed 2022-02-18 by meteyou