mainsail-crew/mainsail #551: Weird behavior on machine settings inputs

r3Fuze, 2022-01-19:
A few weird and confusing things I've noticed with the inputs in the `Machine Settings` panel:

- Manually inputting an integer in an input and then pressing the increase arrow to the right adds the step value of the input to the entered integer, for example putting `300` in the input and pressing the up arrow changes the value to `3005`. This happens on all inputs that use whole numbers.

- Manually inputting a float in an input that accepts it and then pressing the up arrow completely erases the value in the input. Inputting an integer works as above.

- Inputting a value in an input and then unfocusing the input without pressing enter does not make it clear that you haven't actually changed the setting unless you look at the console output.

- The pressure advance input is limited to 1 even though the advance value can be above 1 in Klipper.

I thought some of these problems were caused by #537 but they were also broken on the commit before the PR was merged.

th33xitus, 2022-01-19:
I was able to reproduce point 1 and 2. That is indeed not an intended behavior and has to be fixed. Thanks for pointing that out.

> Inputting a value in an input and then unfocusing the input without pressing enter does not make it clear that you haven't actually changed the setting unless you look at the console output.

That is currently the intended behavior. Manual input has to be confirmed by pressing enter. Where the arrows directly send the command.

> The pressure advance input is limited to 1 even though the advance value can be above 1 in Klipper.

That is correct. Also the smooth time input field has an upper limit of 0.2
Do you have a valid reason why the upper limits should be removed? A possibly required pressure advance value greater than 1 is an indicator that pressure advance in general doesn't seem to have any impact at all on the print quality.
Quoting the official Klipper documentation and configuration reference here:
> Typical pressure advance values are between 0.050 and 1.000 (the high end usually only with bowden extruders). If there is no significant improvement with a pressure advance up to 1.000, then pressure advance is unlikely to improve the quality of prints. Return to a default configuration with pressure advance disabled.

```
#pressure_advance_smooth_time: 0.040
#   A time range (in seconds) to use when calculating the average
#   extruder velocity for pressure advance. A larger value results in
#   smoother extruder movements. This parameter may not exceed 200ms.
#   This setting only applies if pressure_advance is non-zero. The
#   default is 0.040 (40 milliseconds).
```

https://github.com/Klipper3d/klipper/blob/master/docs/Config_Reference.md#extruder
https://github.com/Klipper3d/klipper/blob/master/docs/Pressure_Advance.md

r3Fuze, 2022-01-19:
> Do you have a valid reason why the upper limits should be removed?

In my opinion it doesn't make sense that the interface puts a hard limit on the value when there isn't a hard limit in Klipper.
`1.146` was what gave me the best result when tuning. I could probably drop my PA to `1.000` without a noticeable drop in quality but it feels a little weird to change the settings of my printer because the interface (not Klipper) is complaining 😄 

[image]

th33xitus, 2022-01-19:
@r3Fuze if you feel "adventurous" you can download and test the automatically generated artifact:

r3Fuze, 2022-01-19:
I've tested the changes and everything works great now 👍

th33xitus, 2022-01-19:
Thank you for your feedback! 😀

PR #555 (2022-01-22): fix: input field and spinner bug (fixes #551)
* fix: reset unconfirmed input on blur (fixes #551)
* fix: remove pressure advance upper limit
commit 75aad8b8 (2022-01-22): fix: input field and spinner bug (fixes #551) (#555)