mainsail-crew/mainsail #566: Add a Percent Target for temperature chart [FR]

arne182, 2022-01-22:
The idea is that instead of having a few different temperature targets on a graph that is going from 0°C to max °C. 

You could rather have it show the target temperature as 1.0 and vary the scale to show the amount of variation there is to the Target temperature. So that would mean instead of 
[image]
You have the two curves much closer together and are able to see how well the PID is for your heater.

arne182, 2022-01-22:
[image]
Something like this. Which shows much clearer that the heater bed has times where it is 1.5°C over the target temperature.

ManuelMuehlberger, 2026-10-15:
Although this is an older issue I still think this would be a nice feature. I would love to implement this as an option in the temperature chart settings.

meteyou, 2026-10-15:
@ManuelMuehlberger this would be very nice! I just hadn't the time to implement it until now.