mainsail-crew/mainsail #574: No thumbnail preview

Aussie84d, 2022-01-24:
**Describe the bug:**\
No thumbnail preview on MainSail, large (400x300) or small (32x32) 

**Expected behavior:**\
Thumbnails to be visible on the GCode file selection and on the dashboard

**Screenshots:**\
[image]
[image]
[image]

**Logfiles:**\
[3DBenchy - 0.2mm, 10 (PLA).txt](https://github.com/mainsail-crew/mainsail/files/7921941/3DBenchy.-.0.2mm.10.PLA.txt)
[moonraker.log](https://github.com/mainsail-crew/mainsail/files/7921945/moonraker.log)

th33xitus, 2022-01-24:
Hi, we have to investigate that issue and track it down. 

But anyways, i pinned down the issue and its the `,` in the filename. Please remove it.
A general advice and sidenote: For best practices, keep special characters out of any filenames. Always 😄

ngynkvn, 2022-01-24:
The issue might be related to usage of [`encodeURI`](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/encodeURI) rather than [`encodeURIComponent`](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/encodeURIComponent) in GET requests to moonraker.

meteyou, 2022-01-24:
i think you will get problems with sub directories and `encodeURIComponent`

meteyou, 2022-01-28:
ok. we found a solution for this issue. there was a small bug in moonraker. Arksine fixed it here: https://github.com/arksine/moonraker/commit/04477705b07f2fefd7666c7d3159d5e9ee17ee47

closed 2022-01-28 by meteyou