mainsail-crew/mainsail #576: [FR]

afbenevides, 2022-01-26:
In a multi printer setup in mainsail on 2.1 it is not possible to specify the viewing order of all the printers in the multi-printers view.

**Describe the solution you'd like**
In the printers panel, where you input the adresses of all printer, there could be a way to reorder the printers like in a kanban style, or directly in the prompt, there could be a field to insert the number of order where we want the printer to be shown in the multi printers view.

**Describe alternatives you've considered**
I think we can delete all and re-enter them in the order we want them but not sure

**Additional context**
2.1 is really great!

meteyou, 2022-01-26:
thx for the request! I personally miss this function every time I change the printer location...

afbenevides, 2022-01-26:
I couldn't agree more!

WayneManion, 2023-07-19:
I would really like to be able to go directly to a specific printer by URL. Or go to a specific type of page. 

Example:
http://mainsail.local/oneSpecificPrinter
http://mainsail.local/aDifferentPrinter

Going really nuts:
http://mainsail.local/oneSpecificPrinter/console

Keep up the great work!

meteyou, 2023-07-19:
@WayneManion this have nothing todo with this FR itself, and something like that will not be possible. if you want to have different urls for different printers, you have to use different mainsail instances.