mainsail-crew/mainsail #577: Can't delete printer form multi printer view

afbenevides, 2022-01-26:
**Describe the bug:**\
On mainsail 2.1, in the multi-printers view config panel, if I delete all printer (to reorder) and add them in the good order, when I refresh the page I get all the previous order + the new ones, so printers are duplicated. I think the delete button, deletes it in the mainsail panel, but somehow the last settings are kept on top of the new ones.

**Expected behavior:**\
When I delete a printer in the config panel of multi printers, I expect it to not reappear after

afbenevides, 2022-01-27:
As an update on this topic, I could delete all multi-printers via the factory reset option. I only reset the printers. Then I can add them back in the order I want and I don't have duplicates anymore. The delete function still does not work but this workaround does.

pataar, 2022-01-27:
I fixed this in #579.