mainsail-crew/mainsail #597: [FR] Include heater power output in heater graph

hapklaar, 2022-01-29:
I'm checking out several heater/thermistor combinations and would like to be able to see the amount of power delivered to the heater in the graph. Fluidd has this, but I recently switched to Mainsail and really miss this insight.

Please add the feature to Mainsail to include current heater power in the heater graph, next to set temperature and current temperature.

Alternative would be to go back to Fluidd, but that's not ideal.

th33xitus, 2022-01-29:
Click on the heater name and check the checkbox you can see below:

[image]

hapklaar, 2022-01-29:
Oh great, didn't know that option existed there. Thanks!

closed 2022-01-29 by hapklaar