mainsail-crew/mainsail #603: [FR]Multi-Printers command

afbenevides, 2022-02-02:
**Describe the solution you'd like**
In the multi-printers view, it would be interesting to have some multi-printers commands. Example stop all hosts or update all on all hosts

**Describe alternatives you've considered**
Could that be done on a macro on one master printer that would command the other printers, like in the new macro management, have a remote printers macros group that do that?

**Additional context**
For the update all context is that we have to individually trigger all other printers to update. (I could probably do a bash script that does that... but in mainsail would be awesome)

For the stop/halt all hosts, before putting power off on the main power bar of the multi-printers rack, I like to close all Pi. Another thing is that for this particular one, maybe we would need remote printers command groups? There may be someone with more then one power bar / multi-printer group, and these different groups could be managed separately? Not my case, so a all printers would be enough for most I guess?

lixxbox, 2022-02-08:
Hi there,
the team has to put their heads together to think about this. At the moment, we are not sure what is doable and how it can be implemented in a suitable way.

If i understand correctly, you would like to have the ability in the printer overview to:
- stop all hosts (power off? emergency stop? cancel print?)
- update all (klipper, mainsail, moonraker, system, timelapse, ....) on all hosts
- a macro panel, whose macros run on all printers?

my _personal_ opinion on this: 
i'm not sure about it. as it becomes difficult to output feedback of individual actions and then errors might be overseen or it might become difficult to debug things, etc.

afbenevides, 2022-02-08:
I think the most usefull commands/features for multiple printers would be:

1- update all 
2- power off all hosts

the grouping of the printers, question still remains but could be adressed in another iteration. For now different mainsail instance for each group of printers would do it.

I think other commands like the macros and all could be usefull, but in a secondary level.

what do you think?